thien94/vision_to_mavros #6: Is it possible to use vision_to_mavros repo for usb_cam_node?

hughhugh, 2019-09-16:
Hi, @hoangthien94 
I am using a Logitech camera to identify the ARUCO and estimate the pose. The problem is I do not have a T265 with me  right now. So is the vision_to_mavros repo compatible with the usb_cam_node? I can get the image and recognize the pose.  Can I use it to transform the cam_pose by this node? Thank you.

hoangthien94, 2019-09-16:
Hi @hughhugh,

The ROS nodes that are responsible for camera driver (`usb_cam`) and fiducial tag detection (Aruco) are not part of the package, as the `vision_to_mavros` node only reads from `/tf` topic, processes data, then re-publishes a new topic that is compatible with `mavros`. In other words, the choice of camera and tag is independent of this node, as long as the data contained in `/tf` is correct.

Specifically for the case of Aruco tag, there is already an  [ArduPilot wiki page](http://ardupilot.org/dev/docs/ros-aruco-detection.html) and [blog post](https://discuss.ardupilot.org/t/indoor-autonomous-flight-with-arducopter-ros-and-aruco-boards-detection/34699) that explains in details what need to be done, so check them out.

hughhugh, 2019-09-16:
@hoangthien94 
I have tried the  April tag today. I can `rostopic echo /tf`， but I can’t echo the /mavros/vision_pose/pose. I have put the tag board in front of the camera. Thanks.

hoangthien94, 2019-09-17:
May I suggest that you echo the `/tf` topic and figure out the `source_frame_id` and `target_frame_id` from your aruco tag first, replace the respective params in the launch file or in the code, then try again and see how it goes.

hughhugh, 2019-09-17:
@hoangthien94 
Hi, I am using Apiltag to test the vision_to_mavros according to the Ardupilot Apriltag wiki.

hoangthien94, 2019-09-17:
Glad it works out for you. The RViz should work as described in the wiki.

hughhugh, 2019-09-17:
@hoangthien94 
I have read the src code, it seems that the position of the camera just equals to the position_orig if the camera aligns to the world frame. The `QuaternionFromRPY` is first along the` x axis`, second the` y axis`, finally rotate along the `z axis`. The orientation of the camera is  saved in the` quat_body`. In my case, I combine the apriltag with ardurover firmware. Maybe I have to modify the src code. Thank you.

closed 2019-09-27 by hoangthien94