thien94/vision_to_mavros #10: No message VISION_POSE_ESTIMATE seen at APM Planner

dimaxano, 2019-11-21:
Hello, Thien!
Thank you very much for your work! I really appreciate it.

I have the next problem: i'm running mavros node and publish `/mavros/vision_pose/pose` topic but I do not see it in APM Planner's MAVLink Inspector.
Also I am sending VISION_POSE_ESTIMATE through DroneKit. The same result.

Did you met such problems? Any ideas what's is going wrong?

thien94, 2019-11-21:
Hi @dimaxano , have you tried the other ground control system such as Mission Planner or QGroundControl? I think that would be a good first step to rule out whether the issue is on the GCS side or the flight controller side?

If you are using radio telemetry, you might want to also try again with direct USB cable to the GCS computer. Sometimes the bandwidth of the telemetry does not accommodate the vision topic.

dimaxano, 2019-11-21:
Everything is running in SITL
And yeah, I am installing QGround now. Will see if it helps

dimaxano, 2019-11-21:
Unfortunately, there is no VISION_POSE_ESTIMATE message neither in APM Planner nor QGroundControl.
I am using the next peace of code to send message via MAVLink
https://gist.github.com/dimaxano/8d8a086b64c2f4a337c1eb5330525164

Also interesting fact: when I start sending messages from my host computer i see STATUSTEXT message in MAVLink Inspector with text "EKF IMU initial pos NED = 1,2,3 (m)" . `1,2,3` are values I am sending from my script.

thien94, 2019-11-21:
The issue might be related to settings in `mavros`. I suggest you check out this discussion and see if anything comes up useful: https://discuss.ardupilot.org/t/vision-position-estimate-not-appearing-in-qgroundcontrol/23978

erikasnow, 2019-11-21:
@dimaxano Are you using the vision_to_mavros node, or are you trying to publish the message in another way?

dimaxano, 2019-11-22:
I've tried as sending messages directly to MAVLink (via dronekit, like [this)](https://github.com/thien94/vision_to_mavros/blob/dcd2d0cc782b133751ffca21d3fa737f636e9b92/scripts/t265_to_mavlink.py#L185) as via mavros (just publishing messages to `/mavros/vision_pose/pose`)
In both cases that ardupilot changes its local position to values sent by me, but I still do not have VISION_POSE_ESTIMATE message in GCS. I've tested in APM Plannes and QGC.
Also I set up parameters proposed in this topic https://discuss.ardupilot.org/t/vision-position-estimate-not-appearing-in-qgroundcontrol/23978

thien94, 2019-11-23:
@dimaxano 
Following this wiki for [SITL Advanced Testing for visual positioning](http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#testing-visual-positioning), I can see the `VISION_POSITION_ESTIMATE` message on GCS, so I suggest you take a look at this example as well. Note that although the name is `vicon`, the underlying messages are vision (see [here](https://github.com/ArduPilot/ardupilot/pull/11809))

I see that you are testing in SITL. Have you also tried with a physical flight controller? The settings and params are sometimes quite different between the two, maybe that would help.

dimaxano, 2019-11-25:
Thank you very much for your advice! I'll try it today and report how situation is going

dimaxano, 2019-12-02:
Hey!
Unfortunately, for now I have no possibility to connect FCU with companion computer by micro-USB and serial simultaneously (one for GCS, another for mavlink2mavros communication). I will reopen this issue, if I fix it.

closed 2019-12-02 by dimaxano

FPSychotic, 2023-10-31:
I have this same issue with real FCU, Kakutef7 with custom built firmaware, I also receive the same messages of IMU0 using external, it detects when I run cartographer or t263,I see the car, and maybe even works, not sure how check that, but the message doesn,t appear, @dimaxano ,I know this was time ago, but could you or you remember how you solved it if so. Thanks.