thien94/vision_to_mavros #13: ros melodic catkin_make error

skyaid, 2020-02-14:
ros: melodic
computer: jetson nano

[ 90%] Building CXX object vision_to_mavros-master/CMakeFiles/t265_fisheye_undistort_node.dir/src/t265_fisheye_undistort.cpp.o
/home/nvidia/catkin_ws/src/vision_to_mavros-master/src/t265_fisheye_undistort.cpp: In function ‘void init_rectification_map(std::__cxx11::string)’:
/home/nvidia/catkin_ws/src/vision_to_mavros-master/src/t265_fisheye_undistort.cpp:61:17: error: ‘CV_CALIB_ZERO_DISPARITY’ was not declared in this scope
                 CV_CALIB_ZERO_DISPARITY,

thien94, 2020-02-14:
It seems this is an OpenCV issue with ROS Melodic, which I have not tested with. Can you try to install the cv-related packages again and see if it helps?
```
sudo apt-get install ros-(ROS version name)-cv-bridge
sudo apt-get install ros-(ROS version name)-vision-opencv
```

thien94, 2020-03-03:
Hi @skyaid , just want to check if the above suggestions worked for you?

onlineth, 2020-03-13:
@thien94 I am running into the same issue.  Unfortunately, even after installing those packages, it still had the same issue.

onlineth, 2020-03-13:
Okay, so modifying the following file `vision_to_mavros/src/t265_fisheye_undistort.cpp` did the trick.  I believe the third line is really the only one that was needed but I had those three in there none the less,

```c++
#include <opencv2/calib3d/calib3d.hpp>
#include <opencv2/calib3d.hpp>
#include <opencv2/calib3d/calib3d_c.h>
```

thien94, 2020-03-13:
Hi @onlineth and @skyaid, I got a Nano and tried it out. Interestingly, I did not encounter the issues that you are facing and the build ran successfully with the repo at it is.
@onlineth May you try installing the opencv and try building it again:
 ```
sudo apt-get install ros-melodic-cv-bridge
sudo apt-get install ros-melodic-vision-opencv
```
or if you already have them installed, update to the latest version:
```
sudo apt-get update && sudo apt-get upgrade
rosdep update
```

onlineth, 2020-03-13:
Actually I compiled mine using the latest release from the official opencv repo.

Btw, have you ever had your drone crash and your RealSense T265 eat concrete?  Any simple solution for it not booting?

thien94, 2020-03-13:
> Actually I compiled mine using the latest release from the official opencv repo.

I guess this is where the problem is, the mismatch between the installation of opencv that I installed and yours. Can you check if `calib3d` is installed as part of the installed opencv? Mine is located under `/usr/include/opencv2/calib3d/`. If not, you can install it via:
```
sudo apt-get update -y
sudo apt-get install -y libopencv-calib3d-dev
```

> have you ever had your drone crash and your RealSense T265 eat concrete? Any simple solution for it not booting

I have had a few near-miss experiences, thankfully I was awared that it could happen and regain control immediately when the drone went haywire. 

If the device can still be recognized by the computer, then there is some hope and I suggest you head over to the official repo for support: https://github.com/IntelRealSense/librealsense. 
If it is bricked and not recognizable anymore, an RMA is more or less the only thing left to do.

skyaid, 2020-03-18:
tahnjs

closed 2020-03-18 by skyaid