thien94/vision_to_mavros #14: Regarding MAVROS error during compile

netgittemp, 2020-03-02:
Hi,
I am writing to ask about error during complie.
I installed MAVROS package long time ago, but the following error showed up.

`/catkin_ws/src/vision_to_mavros/src/vision_to_mavros.cpp:7:39: fatal error: mavros_msgs/LandingTarget.h: No such file or directory
`

Can I ask how you install MAVROS package?
or could you advice me to resolve this issue?

Thank you.

thien94, 2020-03-03:
You can follow the standard binary installation method: https://dev.px4.io/v1.9.0/en/ros/mavros_installation.html#binary-installation-debian--ubuntu

netgittemp, 2020-03-05:
Thank you for your quick response.

Now I am doing "Verify that each ROS node is working"

> There are 3 nodes running in this setup. Launch in 3 separated terminals on RPi:
> 
>     T265 node: roslaunch realsense2_camera rs_t265.launch
>         The topic /camera/odom/sample/ and /tf should be published at 200Hz.
>     MAVROS node: roslaunch mavros apm.launch (with fcu_url and other parameters in apm.launch modified accordingly).
>         rostopic echo /mavros/state should show that FCU is connected.
>         rostopic echo /mavros/vision_pose/pose is not published
>     vision_to_mavros node: roslaunch vision_to_mavros t265_tf_to_mavros.launch
>         rostopic echo /mavros/vision_pose/pose should now show pose data from the T265.
>         rostopic hz /mavros/vision_pose/pose should show that the topic is being published at 30Hz.
> 

Here, how can I check the rate?

thien94, 2020-03-06:
Hi @netgittemp , you can check the rate by issuing the command:
```
rostopic hz /camera/odom/sample/      %==> 200Hz
rostopic hz /mavros/vision_pose/pose  %==> 30Hz
```

thien94, 2020-05-06:
If the problem persists, please don't hesitate to reopen the issue.

closed 2020-05-06 by thien94