thien94/vision_to_mavros #15: ArduPilot integration enhancements

rmackay9, 2020-04-04:
Here are some ideas to improve integration with ArduPilot:

- [ ] status text messages sent to the autopilot when:
 - the script starts.  Something like "T265: connecting to camera.."
 - the script succeeds or fails connecting to the camera, "T265: connected to camera" or "T265: failed to connect to camera"
 - the scripts lose connection with the camera (i.e. the user unplugs the camera).
- [ ] pass reset information from the camera to AP using vision-position-estimate's "reset_counter" mavlink2 extension field.
- [ ] send confidence values from the camera to AP using vision-position-estimate's "covariance" mavlink2 extension field?  I don't know how we would do the translation.. I'm actually not even sure it matters.
- [ ] send [vision-position-delta](https://mavlink.io/en/messages/ardupilotmega.html#VISION_POSITION_DELTA) messages to allow easier integration with GPS

thien94, 2020-04-04:
Hi @rmackay9 , thank you for your comment. This list of improvements is very helpful and I will definitely get to it soon.

I think before further development, I should make clear the distinctions between the script `t265_to_mavlink.py`  in this repo and the one in [ArduPilot/companion](https://github.com/ArduPilot/companion) repo:
- The one in this repo ([vision_to_mavros](https://github.com/thien94/vision_to_mavros)) needs to be consistent with what the descriptions in the [GSoC 2019 blog posts](https://discuss.ardupilot.org/t/gsoc-2019-integration-of-ardupilot-and-vio-tracking-camera-for-gps-less-localization-and-navigation/42394) and [Ardupilot wiki](https://ardupilot.org/copter/docs/common-vio-tracking-camera.html). Right now they are in sync, so I tend to keep it that way. For example, the option `--auto_set_ekf_home_enable` is `True` by default since that was what is written down in the blog post.
- Moving forward, I think the new developments that give benefits directly to ArduPilot should be done on [the `t265_to_mavlink.py` in the companion repo](https://github.com/ArduPilot/companion/blob/master/Common/Ubuntu/librealsense/t265_to_mavlink.py to avoid confusion and extra efforts to work on different repos and then bring the changes from one place to another. If you agree with this idea, I shall move the new changes to the [companion repo](https://github.com/ArduPilot/companion) instead. Henceforth, the wikis page will be updated to the script in companion repo, while the GSoc 2019 blog posts remain in sync with what was written in this repo (also, updating the blog posts are not something I can do, so it is also better this way).

How do you think?

rmackay9, 2020-04-08:
Hi @thien94,

I looked more closely at the apsync scripts recently and found that although there is a copy of the t265_to_mavlink.py script there, it's not being used! :-).  Instead it downloads and then copies into place the version in this repo.  I think we should move that script somewhere but I'm not too worried about it.

I don't personally think you should worry too much about moving on from what's written in the GSoC 2019 blog posts.  Software has to move forward and the info in blog posts becomes out of date.

Anyway, more importantly thanks for working on some of the issues above!

thien94, 2020-04-09:
@rmackay9 In that case I will focus on the development in this repo.
I am adding support for `VISION_POSITION_DELTA` message into this branch https://github.com/thien94/vision_to_mavros/commits/AP_enhancement_01 and hopefully it will be merged into master soon. Then I would love to have your help in doing some flight tests.

rmackay9, 2020-04-10:
@thien94, by the way, I noticed that the sensor's default position in the scripts is 0.05 (i.e. 5cm) forward.  We can handle this on the AP side now so it would be good if this could be defaulted to zero somehow.  I've added this to the list above.

rmackay9, 2020-04-11:
By the way, I've found another issue related to the T265 scripts sending status to the ground station.  It seems that if the RPI/T265 are on the "far side" of the telemetry connection between the autopilot and the GCS (i.e. T265/RPI <-> AutoPilot <-> Ground Station) then the send-text messages from the T265/RPI are not sent to the ground station.

If setup is like this: AutoPilot <-> T265/RPI <-> GCS (i.e. the RPI4's wifi access point is used for telemetry) then the messages are sent.

I've created an enhancement request for ArduPilot here: https://github.com/ArduPilot/ardupilot/issues/14055

thien94, 2020-04-11:
@rmackay9 Yes, I also experienced the dropped messages issue to GCS from time to time. As it relates to communication, I think the available bandwidth at the moment the message is sent plays an important role here. I think if the script tries to send a T265 message at the same time, or very close to, when the FCU is sending another message, it's likely the T265 message would be dropped regardless of the communication type, as can be seen here https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L1407.

Some ideas for potential solutions:
1. Having some kind of feedback from the FCU that the latest request for statustext message has been sent, 
2. Having a queue for statustext message on the script side making sure all requests are sent, but this would require that we have some kind of flag indicating the message has been sent.

rmackay9, 2020-07-20:
I think we can close this now, thanks!

closed 2020-07-20 by rmackay9