thien94/vision_to_mavros #18: T265 linear velocity to /mavros/vision_speed/speed_twist_cov

weiLiangHare, 2020-06-14:
T265 also provides linear velocity measurements, can It  be used to improve the state estimate.

thien94, 2020-06-14:
That is certainly possible. However, there are some issues raised on the `librealsense` repos regarding the instability of the speed data from the T265, so it should be used with caution during development (example: https://github.com/IntelRealSense/librealsense/issues/4876, https://github.com/IntelRealSense/librealsense/issues/6138).

On the other hand, ArduPilot only added support for speed estimate recently https://github.com/ArduPilot/ardupilot/pull/14404, so the current implementation of this repo does not use the speed data yet.

weiLiangHare, 2020-06-14:
Thank you for your quick reply，so It is still possible improve the state estimate。

thien94, 2020-06-14:
No problems. Thanks for the discussion.

closed 2020-06-14 by thien94