thien94/vision_to_mavros #20: Why no covariance information in Mavlink? How to configure if my ZED 2 camera is facing back?

cnpcshangbo, 2020-07-07:
Hi,
I am able to get Vision_position_estimate message transferred to Pixhawk.
[image]
Is the covariance information supposed to be always zero?

I found [this file](https://github.com/thien94/vision_to_mavros/blob/050163c40832cc4987cb34b4a8b4abd3ff6a4309/launch/t265_tf_to_mavros.launch#L12-L23) contains forward facing and downward facing configuration.
Do you have some suggestions for backward facing?
[image]
Thanks,
Bo

thien94, 2020-07-08:
Hi Bo, thank you for checking out the code.

> Is the covariance information supposed to be always zero?

If you are using the latest [t265_to_mavlink.py](https://github.com/thien94/vision_to_mavros/blob/master/scripts/t265_to_mavlink.py) Python script, the covariance should be filled. If you are using the ROS node, it is not at the moment (not yet implemented). The covariance is computed from the `tracker_confidence` data provided by the T265.

> Do you have some suggestions for backward facing?

I think changing `yaw_cam = -1.5707963` (180 degrees), roll and pitch = 0, should work.

cnpcshangbo, 2020-07-10:
https://drive.google.com/file/d/1Iz1zONpoX9bpuCF1f5Kmn7V84Ev_tI2F/view?usp=drivesdk
When I switched to Loiter, the drone went down to the ground. When I tried to take off, the drone crashed to its left. It seems there’s something wrong with coordination transfer. I think the red arrow should always go with the axis. I’ll do more tests on Monday.

thien94, 2020-07-11:
In my experience, flight test can be done later to avoid crashes. Handheld test is sufficient to verify the data processing.

The most important part is to figure out the coordinate and transformation correctly. This [reference](https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html#ros_reference_frames) sums it up very well.
- First, you should verify the coordinate frame of the position data from the ZED 2 camera. Make sure it is in ROS coordinate.
- Next, check that the output of `vision_to_mavros` fits the coordinate of of `mavros` (also in ROS).
- Then check with Mission Planner of QGC that the `VISION_POSITION_ESTIMATE` received by ArduPilot is in NED.
- If all goes well, the local position estimate should be quite stable even if you handheld the copter and move it around very aggressively.

cnpcshangbo, 2020-07-11:
Thanks for your information.
I noticed ZED 2 thinks it faces East when it starts. Does Real Sense assume it faces North when it starts?
Do you always boot your drone when it faces North?
Thanks,
Bo

thien94, 2020-07-13:
> Do you always boot your drone when it faces North?

That is the default setting to make the heading 0 (north), not that the drone is actually facing north. If you have compass enabled, however, then the heading will have some relation with earth's north.

cnpcshangbo, 2020-07-14:
Hi @thien94 ,
I have changed the ZED 2 to face front now. I have done a ground test. It looks good. However, the drone still wants to crash on its right when I do flight test. https://drive.google.com/file/d/1jdq2rFZqkrTkqZfhMXIzicfx5rpZhXGv/view
Do you think the ground test looks good?
Do you think this issue is related to my ZED positional tracking update rate? The rate is 15 Hz now.
Thanks.
@cnpcshangbo

cnpcshangbo, 2020-07-14:
Hi @thien94 ,
I tried to change the positional tracking frequency to 30 Hz, however, the drone still tries to crash on the right.
https://youtu.be/dK6mZZX8Pfs
I also noticed that there's an error message: 
[ERROR] [1594752881.672783543]: FCU: PreArm: DCM Roll/Pitch inconsistent by 65 deg
Do you think this is related to there aren't enough visual features in front of the drone?
Also, there's a warning.
[ WARN] [1594752942.548855616]: TM : RTT too high for timesync: 1202.75 ms.
Do you think the high RTT can cause some issue?
It seems I need to check the mission planner data log. Do you have some ideas on which variables should I check?
Thanks,
@cnpcshangbo

thien94, 2020-07-15:
From what I can see, if the position is displayed correctly then the problem is somewhere else.
Was the flight done mostly in Stabilized mode? The position tracking data only works with Loiter mode, so if it is unstable in Stabilize, that means the frame is unbalanced.

I would suggest doing some flight tests in Stabilize (no position data) to confirm whether the crashing to the right is due to the frame itself. You can also skip stabilize and use attitude hold as the beginning, then switch to Loiter. You can also takeoff from Loiter directly, but beware the behavior might be more aggressive depends on the PID tunning.

Regarding the RTT error, you can basically ignore it or suppress it following this discussion: https://discuss.ardupilot.org/t/rtt-too-high-for-timesync-with-sitl-mavros/38224/8

cnpcshangbo, 2020-07-28:
@thien94 Thanks for you help. The main reason for the drone crashing to the right after switching to Loiter mode is I pushed the right lever when I arm the drone. The correct method should be only pushing left lever to arm.

closed 2020-07-28 by cnpcshangbo

DiegoHerrera1890, 2022-09-28:
Hello guys thank you for such amazing discussion.

I have been using t265 but since it is broke I had to change to the zed mini.
I have implemented a scrip to publish the pose and pose with covariance to mavros/vision_pose/...
In the mavlink inspector I can only see VISION_POSITION_ESTIMATE values but before when I used the t265 mavlink inspector was showing odometry data pose and orientation.
Is it okay to just receive VISION_POSITION_ESTIMATE x,y,z,roll,pitch, and yaw?
or What should I do?
[image]

DiegoHerrera1890, 2022-09-28:
> @thien94 Thanks for you help. The main reason for the drone crashing to the right after switching to Loiter mode is I pushed the right lever when I arm the drone. The correct method should be only pushing left lever to arm.

DiegoHerrera1890, 2022-09-28:
> From what I can see, if the position is displayed correctly then the problem is somewhere else. Was the flight done mostly in Stabilized mode? The position tracking data only works with Loiter mode, so if it is unstable in Stabilize, that means the frame is unbalanced.
> 
> I would suggest doing some flight tests in Stabilize (no position data) to confirm whether the crashing to the right is due to the frame itself. You can also skip stabilize and use attitude hold at the beginning, then switch to Loiter. You can also takeoff from Loiter directly, but beware the behavior might be more aggressive depends on the PID tunning.
> 
> Regarding the RTT error, you can basically ignore it or suppress it following this discussion: https://discuss.ardupilot.org/t/rtt-too-high-for-timesync-with-sitl-mavros/38224/8

hichamhendy, 2022-12-21:
That's probably because you're publishing on vision_pose and not mavros/odom/in. Both do the same job. you also don't need to publish on topics when you use vision_pose. It offers another comfortable choice by turning on listen then specify the frames and mavlink will track the camera frame. You can use local position ned in this case to verify.

DiegoHerrera1890, 2023-01-23:
@hichamhendy Thank you so much for your answer. I am sorry for my late reply.
What do you mean when you says `It offers another comfortable choice by turning on listen then specify the frames and mavlink will track the camera frame`

DiegoHerrera1890, 2023-02-07:
thank you for your answer I have been able to publish odometry to mavros/odometry/out.
I have a question, using the t265 camera the covariance was almost zero [0.1 0 0 0 0 0 ...] but using the zed mini the covariance is something like his [2.56478e-07, -6.95478e-08, -2.4564e-08, ...] Is this a proper value?
Also, the twist/linear and twist/angular are all zero... which means it is not publishing velocity, do I have to publish this linear/angular velocity from another node of the zed camera?
Thank you in advance.

DiegoHerrera1890, 2023-09-20:
@egorfolley  I had to switch to the T265 camera because it's easier. But this week I will be using the zed camera and I will let you know how it went. PX4 has a basic guide on the topics required to generate odometry.