thien94/vision_to_mavros #34: Workable when using PX4 as autopilot?

trojancai, 2021-02-08:
Hello, I'm a novice at ros programming for UAV and I'm really interested in this project. However, the autopilot I'm using now is PX4, not Ardupilot as you did; so I wonder if this package could be utilized directly in my ROS workspace without any modification, or could you please give me some advice about that?

Thank you in advance.

thien94, 2021-02-09:
Hi @trojancai , thank you for your interest in the project.

In the case of PX4, more suitable for your requirement would be this package: https://github.com/Auterion/VIO. The detailed instruction can be found here: https://docs.px4.io/master/en/peripherals/camera_t265_vio.html 

Additionally, the PX4 forum would be the place for questions as well as learning from others' problems: https://discuss.px4.io/search?q=T265

Hope this helps.

trojancai, 2021-02-09:
Thanks for your quick reply. I'll take a deeper look.

trojancai, 2021-02-09:
Hi @thien94 , I have some questions about the pose data.

I have learned that package [https://github.com/Auterion/VIO](https://github.com/Auterion/VIO) offers parameter APIs of static transform which can be modified so as to correspond with the translation offset between T265 and the center of flight controller, referring to  [https://docs.px4.io/master/en/peripherals/camera_t265_vio.html](https://docs.px4.io/master/en/peripherals/camera_t265_vio.html).

However, it seems that there is no APIs like above in your package. I wonder if I have missed some code or the pose error caused by the translation offset can actually be neglectable?

thien94, 2021-02-10:
For Ardupilot, the translational offset between the vision sensor and the body frame can be set with [`VISO_POS_XYZ`](https://ardupilot.org/copter/docs/parameters.html#viso-parameters) parameters on the flight controller side.

Initially, the offsets are also taken care of by this package (the legacy code [can still be seen the python version](https://github.com/thien94/vision_to_mavros/blob/master/scripts/t265_to_mavlink.py#L85-L90)). However, it was decided later on that this package should work such that from ArduPilot's perspective, the T265 is a plug-and-play sensor like many other (you plug it in, run the script/package - which can be automated - and data come out) and everything else is handled by ArduPilot (temporal/spatial offset, reliability of data, to name a few).

trojancai, 2021-02-10:
I see. Thank you so much for your detailed answer.

closed 2021-02-10 by trojancai