thien94/vision_to_mavros #35: Problem When starting t265_tf_to_mavros.launch

i-jelly, 2021-04-06:
I`ve got a WARN at the last step of installation.
after launch the t265_tf_to_mavros.launch, it shows that

Get output_rate parameter: 20.000000
[ INFO] [1617727957.134085485]: Get gamma_world parameter: -1.570796
[ INFO] [1617727957.135297278]: Get roll_cam parameter: 3.141593
[ INFO] [1617727957.136439071]: Get pitch_cam parameter: -1.570796
[ INFO] [1617727957.137552549]: Get yaw_cam parameter: 0.000000
[ INFO] [1617727957.138694787]: Precision landing: enabled
[ INFO] [1617727957.139966913]: Get precland_target_frame_id parameter: /landing_target
[ INFO] [1617727957.141412520]: Get precland_camera_frame_id parameter: /camera_fisheye2_optical_frame
[ WARN] [1617727960.154184135]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1617727961.154835400]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. 

and in the MAVLINK Inspector i still can`t see any vision data
i think this problem is related to tf_listener.lookupTransform(target_frame_id, source_frame_id, now, transform); in vision_to_mavros.cpp

thien94, 2021-04-09:
Hi @i-jelly , thank you for the question.

The issue seems to be that there is no "camera_odom_frame" in the current tf tree. If you are using the T265,  you can verify whether it exists or not by echoing the /tf topic and look for the frame with the same name.

i-jelly, 2021-04-15:
Hi @thien94 
Thank you for your attentions,
so wired, the problem happens only when runs on Ubuntu18 even after reinstall the system.
but Ubuntu20 seems to work properly.

ipungg-junior, 2021-04-23:
Still same, i got this.

[ WARN] [1617727961.154835400]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist.

i-jelly, 2021-04-23:
> Still same, i got this.
> 
> [ WARN] [1617727961.154835400]: "camera_link" passed to lookupTransform argument source_frame does not exist.

Hello @ipungg-junior ,  can you share the detail of environment? 
I tried this project on both Ros melodic with Ubuntu18 & Ros noetic with Ubuntu20 at Raspi4 4GRAM,but only Ubuntu20 works fine.

ipungg-junior, 2021-04-23:
Thanks for attention @i-jelly 
It seems topic not exist, like you.
Stressed. :)

thien94, 2021-04-24:
@ipungg-junior 

A quick explanation: the `vision_to_mavros` node receives data from the `tf` tree, particularly from the two frames named `camera_link` defined in`camera_odom_frame` by default. If you are using the [`realsense-ros` driver](https://github.com/IntelRealSense/realsense-ros) for the T265, the names should match by default.

The warning message means that  "camera_link" frame does not exist in the `tf` tree. First you need to check on the `realsense-ros` side: after you launch the `realsense-ros` node, echo the `tf` topic and you should see the frames "camera_link" and "camera_odom_frame". If not, then you have to edit the names of the frames in the [t265_tf_to_mavros.launch](https://github.com/thien94/vision_to_mavros/blob/master/launch/t265_tf_to_mavros.launch) file to match.

Zave1d, 2023-04-06:
Excuse me I want to ask, I am still new to this, what I want to ask is whether to use the t265 real sense camera? Can I use a regular Logitech Camera?
Thank you (:

Zave1d, 2023-04-06:
I have checked (rostopic echo /tf) and can come out the results
While when I try to check using (rostopic echo /masvros/vision_pose/pose) there is no result
 me pusing_-

thien94, 2023-04-06:
Hi @Zave1d , for this project we need a [VIO tracking camera](https://docs.px4.io/main/en/peripherals/camera_t265_vio.html), so unfortunately a regular logitech camera would not suffice.

Zave1d, 2023-04-10:
@thien94 Is that why, when I run the command (rostopic echo /masvros/vision_pose/pose) on the teminal, the drone position data does not come out, while when I run (rostopic echo /tf) it comes out the drone position data
Note: when I hover the tag on the camera

mqgeng, 2024-04-17:
Thanks for your answer. I met the same "“camera_odom_frame” is not exsit" on the ubuntu20.04,  I tried to install the driver as the link: https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy. Then I launched  the node again and there was camera_odom_frame , I am so happy.  Thanks again.

> @ipungg-junior
>
> A quick explanation: the `vision_to_mavros` node receives data from the `tf` tree, particularly from the two frames named `camera_link` defined in`camera_odom_frame` by default. If [...]