thien94/vision_to_mavros #36: PX4的无人机连接T265，无人机起飞时为什么旋转90°。

1506388648, 2021-04-09:
我使用该无人机自主飞行时，貌似T265建立了世界坐标系，每次起飞无人机都必须头朝世界坐标系的x方向，才会进行后面的步骤。

thien94, 2021-04-10:
Hi @1506388648 ,
- Could you post the question in English? It would be very much appreciated.
- If you are using the T265 with PX4 specifically, a good place to visit is the PX4 wiki: https://docs.px4.io/master/en/peripherals/camera_t265_vio.html where they explain everything. For any inquiries, the corresponding Github repo is the place to post questions https://github.com/Auterion/VIO. This repo works with ArduPilot specifically. 
- Regarding your situation, changing `gamma_world` param [in the launch file](https://github.com/thien94/vision_to_mavros/blob/master/launch/t265_tf_to_mavros.launch#L28) might work. Set it to `0` or `+1.5707963` and see if it helps.

1506388648, 2021-04-10:
Thank you very much for the author’s reply, I will pay attention to the translation next time and ask questions.  I will try your method now.

thien94, 2021-11-19:
Feel free to reopen the issue if you need to.

closed 2021-11-19 by thien94