thien94/vision_to_mavros #37: Drone flies well for a while and then it goes forward very quickly while using a backwards facing t265

maxilar20, 2021-04-10:
I'm having an issue where I can't manage to set the yaw variable correctly. The camera is facing backwards and so I set the yaw_camera to 3.1215 but when I'm flying it just wants to go to the side very quickly. It worked when It was just facing forwards but I changed only that value and it doesn't seem to work now.

I also tried giving it a transformation from the t265 frame to a base_link frame that was positioned correctly (and leaving the yaw value at 0) but it only worked for some second until I wanted to move it, then it just flew to the side very quickly.

thien94, 2021-04-10:
Hi @maxilar20, thank you for the question. I have not personally tested the backward-facing configuration, so I can only provide suggestions:
- Is the backward facing camera in normal orientation, i.e. you can read the text on the camera normally and the USB port is to the left of the vehicle? If so, `yaw_cam` should be `3.1415926` while `roll_cam` and `pitch_cam` should be `0`.
- It's recommended to verify the data with handheld test before any real flight.  Move the vehicle in a square. The trajectory shown on the map of Mission Planner should follow the real movements. If it's not, then something is wrong and we need to go back and change something before taking flight.

An example of the trajectory:
[image]

maxilar20, 2021-04-10:
Thanks for the quick reply!

So, I did what you said about the yaw_cam value, and when I try to see the gps position it start with a heading of 270 degrees:
[image]

and when I move it forwards, the drone moves to the south on the map, and when I move it to the right it moves to the west.
[image]
[image]

maxilar20, 2021-04-10:
Is it normal that it's pointing to the south on startup?

thien94, 2021-04-11:
The default heading should be north if `gamma_world` is set correctly. However, do check that the result is repeatable between multiple runs (in case the heading was incorrect due to other reasons not aware of) before testing out a new param.

A bit of a guessing game, but you can record the result with `gamma_world` as 0, +1.570796, -1.570796, 3.1415 to see how the param affects the heading.

closed 2021-04-19 by maxilar20