thien94/vision_to_mavros #41: Question regarding the transformations in ROS setup

hamsterasu, 2021-04-28:
Hi @thien94 ,

In your README, you mentioned 3 frames:
1) source_frame
2) world frame {W}, which is also the `target_frame_id`
3) body frame {B}

and also 4 parameters:
`roll_cam`, `pitch_cam`,`yaw_cam`, `gamma_world`

I have a few questions:

1) What is body frame {B} referring to? Is it a frame that has its origin on the flight controller? 
2) What is the purpose of `gamma_world` parameter if you already have `roll_cam`, `pitch_cam`, `yaw_cam` for the rotation?
3) I have followed your non-ROS setup already and it worked fine for me. Thanks for the great work! In that setup, I noticed that there were these variables in `t265_to_mavlink.py` file to compensate for the camera position offsets. In the ROS setup, I could not find any parameters or variables that is for the same purpose. Is the camera position offset being compensated elsewhere?

thien94, 2021-04-28:
Hi @hamsterasu , many thanks for the questions.
[image]
1. The body frame {B} refers to the robot's body frame, which is (X Forward, Y Right and Z Down). Your assumption is correct.
2. The `gamma_world` is used to rotate the world frame {W} to make sure y-axis is pointing forward at the start. The other parameters `r/p/y_cam` relates to the camera. In a sense,  `gamma_world` is not crucial since without it, the system still works. The only problem is that 0-degree yaw at the beginning will be interpreted as pointing east. With  `gamma_world`, the starting heading will be north.
3. The non-ROS script contains more experimental code than the ROS. Specifically, the camera-body offset can be added. However, at a later point I realized that ArduPilot [already has parameters to set them up](https://ardupilot.org/copter/docs/common-sensor-offset-compensation.html) so they were left out of the ROS part.

hamsterasu, 2021-08-13:
Sorry for the very late reply! I was so caught up in testing my setup after reading your reply that I forgot to reply. Thanks for the clarification, my drone is flying well now :)

closed 2021-08-13 by hamsterasu