thien94/vision_to_mavros #42: Other VISLAM systems configuration

xiaolin198882, 2021-05-11:
What configuration needs to be changed to use other VISLAM systems such as VINS-fusion, thank you!

xiaolin198882, 2021-05-11:
Need some help from you @thien94, thanks

thien94, 2021-05-14:
Hi @xiaolin198882 , sorry for the late reply.

> What configuration needs to be changed to use other VISLAM systems such as VINS-fusion

Firstly, the `target_frame_id` and `source_frame_id` would have to be modified to fit your algorithm's output, assuming that the VISLAM system in question does in fact pump out data in `\tf` tree. Otherwise, you would have to create your own subscriber to obtain the data.
Secondly, the rotation between the frames need to be correct according to the convention of the parameters: https://github.com/thien94/vision_to_mavros/blob/master/launch/t265_tf_to_mavros.launch#L25-L28

> T265 is a bit expensive, so I DIY a cheap vislam device, and there is another question, what is the minimum frequency of vislam output, can it be 10Hz to 30Hz?

Typically 10Hz is the bare minimum frequency to make it usable for other applications, such as navigation and control. The higher the better, of course only when the performance is satisfactory.

closed 2022-03-28 by xiaolin198882