thien94/vision_to_mavros #46: Up2 Apsync and rs_to_mavlink.py

matthewryanburgin191, 2021-07-20:
Good day,

So I've been playing with apsync-up2-d435i-yyyymmdd.tar.xz.
This is the second time playing around with it, this time it doesnt work I've missed something.
ttyS5 does not work, have i missed something?
ACM0 works but when running  rs_to_mavlink.py the hz drop off the obstical distance until 0hz.
This was tested on the orange cube.
Any help would be wounderfull.

Bestregards
Matthew

thien94, 2021-08-02:
Sorry for the late reply.
It is possible a problem with the configuration of the Cube (some parameters might be missing / incorrect?).
Unfortunately I am not very familiar with the Cube variants (except the black version), so a better place to ask would be the forum https://discuss.ardupilot.org/
Hope this helps.

matthewryanburgin191, 2021-08-03:
Hi,

The only way I managed to get around it was to reinstall Ubuntu and start again as the issue seamed to be with the physically port on the development board as the fcu had been tested previously and was functional.

Best regards

Matthew

closed 2021-08-03 by matthewryanburgin191