thien94/vision_to_mavros #50: Error when running roslaunch vision_to_mavros t265_tf_to_mavros.launch

Wina30, 2021-11-05:
I get warning on my terminal, that say "[ WARN] [1636146132.931182478]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. ", that make problem with my drone because when I opened the inspector mavlink, that say is no viso_estimate was connect on my pixhawk.
[image]

thien94, 2021-11-06:
As the error message says, the issue is that there is no `camera_odom_frame` in the `/tf` tree. Either the camera driver is not running, or the name of the frame(s) does not match.

First, check the camera driver launch file, verify that it is working properly and verify the names of the frame ids.
If the driver is running ok, either modify the frame id on the driver's side or in the `t265_tf_to_mavros.launch` file.

Hope this helps.

Wina30, 2021-11-06:
in my rostopic echo /tf doesn't appear any information 
[image]
so is my problem relate to the camera driver?

thien94, 2021-11-07:
Yes, that seems to be the case. If the driver is running properly, the `/tf` should be populated.

Wina30, 2021-11-08:
Thanks about your answer, my realsense was work now, I just reinstall my realsense driver

closed 2021-11-08 by Wina30