thien94/vision_to_mavros #54: VISION TO MAVROS USING D435I CAMERA AND VIO

bhaskar-glitch, 2021-11-25:
Hi @thien94 @mzahana @peterbarker @muramura ,
I want use this vision to mavros script with my D435I camera and VIO(VINS_FUSION).
What changes i have to make to do so? Please help

thien94, 2021-11-25:
I suggest separating the process into a few steps:
1. Being able to run VINS-Fusion using the D435i. You can adapt the [available config files](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/tree/master/config/realsense_d435i) to your sensor. You should have stable, high frequency odometry output.
2. Adapt the launch file [`t265_tf_to_mavros`](https://github.com/thien94/vision_to_mavros/blob/master/launch/t265_tf_to_mavros.launch) to your system. First, all the arguments (`tf` frames name) must be correct. You should have the mavros messages and FCU receiving data, but the data might be in wrong orientation.
3. Second, the parameters need adjustment to correct the data. This one might be a bit complicated if you are not familiar with the concept of frame transformation.
 
Once you reach step 3, we can continue the discussion for the last part.

bhaskar-glitch, 2025-04-04:
> Hello, I am currently working on a D435i SLAM project. I am using RTAB-Map to obtain odometry data and send it to the /mavros/vision_pose/pose topic. The drone does receive the data and can be unlocked, but the takeoff fails. I tried using t265_tf_to_mavros and modifying the frame_id, but the takeoff issue still persists. Could you help me?

I don't know what hardware you are using to do so and how your setup is, I will suggest to use software controller for any kind of movement, also in the case of controller you don't have to worry about vision_pose topic or any other topic of mavros, the controller will itself manage all the things. If i talk about your issue it could be the issue of wrong orientation or noise in the data.

ZephyrKeXiner, 2025-07-24:
Hello, I also want change the T265 to D435i, have you did with this problem?

ZephyrKeXiner, 2025-07-24:
> > Hello, I also want change the T265 to D435i, have you did with this problem?
> 
> Yes, I did with d435i camera and lidar too

Could you tell me how do you modify the code？
I run the set_origin.py but my ardupilot mavros always shows that ”[INFO] [1753355992.993998957]: HP: requesting home position“

ZephyrKeXiner, 2025-07-27:
> > > > Hello, I also want change the T265 to D435i, have you did with this problem?
> > > 
> > > 
> > > Yes, I did with d435i camera and lidar too
> > 
> > 
> > Could you tell me how do you modify the code？ I have no idea how to adapt it to D435i.
> 
> You have to modify the code of vision to mavros to adapt the frame IDs of d435i camera, I can't provide you the exact code but just match the d435i camera frame with vision to mavros and all set

Thank you very much! I have found the way to adapt d435i camera!