thien94/vision_to_mavros #64: I can't get /mavros/vision_pose/pose to publish.

Dawgomatic, 2023-09-09:
Setup: RPI4 | Ubuntu 20.04 | ROS Noetic | UART GPIO pins to CubeOrange Plus | USB webcam

Followed - https://ardupilot.org/dev/docs/ros-apriltag-detection.html

roslaunch usb_cam usb_cam-test.launch - provides video output.

roslaunch mavros apm.launch fcu_url:=<my-tty-port>:<my-baud-rate> - Provides imu data on /mavros/imu/data and true connection status on /mavros/status.

roslaunch vision_to_mavros apriltags_to_mavros.launch - shows data on /tf 

Problem: no data on /mavros/vision_pose/pose and no scene data in RVIZ.

roslaunch usb_cam usb_cam-test.launch output.
```
SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/color_format: yuv422p
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/video_device: /dev/video0

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311/

process[usb_cam-1]: started with pid [2479]
process[image_view-2]: started with pid [2480]
[ INFO] [1694118679.554671404]: Initializing nodelet with 4 worker threads.
[ INFO] [1694118680.331101128]: Using transport "raw"
[ INFO] [1694118680.592663318]: using default calibration URL
[ INFO] [1694118680.603234099]: camera calibration URL: file:///home/ubuntu/.ros/camera_info/head_camera.yaml
[ INFO] [1694118680.608789041]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS
```
roslaunch vision_to_mavros apriltags_to_mavros.launch output.
```
SUMMARY
========

CLEAR PARAMETERS
 * /apriltag_ros_continuous_node/

PARAMETERS
 * /apriltag_ros_continuous_node/camera_frame: camera_frame
 * /apriltag_ros_continuous_node/max_hamming_dist: 2
 * /apriltag_ros_continuous_node/publish_tag_detections_image: True
 * /apriltag_ros_continuous_node/publish_tf: True
 * /apriltag_ros_continuous_node/standalone_tags: []
 * /apriltag_ros_continuous_node/tag_blur: 0.0
 * /apriltag_ros_continuous_node/tag_bundles: [{'name': 'a3_bun...
 * /apriltag_ros_continuous_node/tag_debug: 0
 * /apriltag_ros_continuous_node/tag_decimate: 1.0
 * /apriltag_ros_continuous_node/tag_family: tag36h11
 * /apriltag_ros_continuous_node/tag_refine_edges: 1
 * /apriltag_ros_continuous_node/tag_threads: 2
 * /apriltag_ros_continuous_node/transport_hint: raw
 * /apriltags_to_mavros/output_rate: 30
 * /apriltags_to_mavros/source_frame_id: camera
 * /apriltags_to_mavros/target_frame_id: A3_bundle
 * /camera/camera_frame_id: camera_frame
 * /camera/camera_info_url: file:///home/ubun...
 * /camera/camera_name: camera
 * /camera/image_height: 480
 * /camera/image_width: 640
 * /camera/io_method: mmap
 * /camera/pixel_format: yuyv
 * /camera/video_device: /dev/video0
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    apriltag_ros_continuous_node (apriltag_ros/apriltag_ros_continuous_node)
    apriltags_to_mavros (vision_to_mavros/vision_to_mavros_node)
    camera (usb_cam/usb_cam_node)
  /camera/
    image_proc (image_proc/image_proc)

ROS_MASTER_URI=http://localhost:11311/

process[camera-1]: started with pid [2339]
process[camera/image_proc-2]: started with pid [2340]
process[apriltag_ros_continuous_node-3]: started with pid [2341]
process[apriltags_to_mavros-4]: started with pid [2342]
[ WARN] [1694118460.817885468]: Using default target_frame_id: /camera_odom_frame
[ WARN] [1694118460.825135741]: Using default source_frame_id: /camera_link
[ WARN] [1694118460.828415109]: Using default output_rate: 20.000000
[ WARN] [1694118460.831151507]: Using default gamma_world: -1.570796
[ WARN] [1694118460.835536015]: Using default roll_cam: 0.000000
[ WARN] [1694118460.841436394]: Using default pitch_cam: 0.000000
[ WARN] [1694118460.843972927]: Using default yaw_cam: 1.570796
[ INFO] [1694118460.846724955]: Precision landing disabled by default
[ INFO] [1694118460.884311920]: Initializing nodelet with 4 worker threads.
[ INFO] [1694118462.216907851]: Loading tag bundle 'a3_bundle'
[ INFO] [1694118462.217198382]:  0) id: 0, size: 0.078, p = [0,0,0], q = [1,0,0,0]
[ INFO] [1694118462.217330101]:  1) id: 1, size: 0.078, p = [0.095,0,0], q = [1,0,0,0]
[ INFO] [1694118462.217407358]:  2) id: 2, size: 0.078, p = [0.191,0,0], q = [1,0,0,0]
[ INFO] [1694118462.217499171]:  3) id: 3, size: 0.078, p = [0,-0.097,0], q = [1,0,0,0]
[ INFO] [1694118462.217593132]:  4) id: 4, size: 0.078, p = [0.095,-0.097,0], q = [1,0,0,0]
[ INFO] [1694118462.217658667]:  5) id: 5, size: 0.078, p = [0.191,-0.097,0], q = [1,0,0,0]
[ INFO] [1694118462.217719481]:  6) id: 6, size: 0.078, p = [0,-0.197,0], q = [1,0,0,0]
[ INFO] [1694118462.217784146]:  7) id: 7, size: 0.078, p = [0.095,-0.197,0], q = [1,0,0,0]
[ INFO] [1694118462.217874162]:  8) id: 8, size: 0.078, p = [0.191,-0.197,0], q = [1,0,0,0]
[ INFO] [1694118462.217944290]:  9) id: 9, size: 0.078, p = [0,-0.295,0], q = [1,0,0,0]
[ INFO] [1694118462.218008104]:  10) id: 10, size: 0.078, p = [0.095,-0.295,0], q = [1,0,0,0]
[ INFO] [1694118462.218066658]:  11) id: 11, size: 0.078, p = [0.191,-0.295,0], q = [1,0,0,0]
[ WARN] [1694118462.219993836]: remove_duplicates parameter not provided. Defaulting to true
[ INFO] [1694118462.557640973]: camera calibration URL: file:///home/ubuntu/ost.yaml
[ INFO] [1694118462.570569601]: Starting 'camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS
[ WARN] [1694118462.888288772]: unknown control 'focus_auto'

[ INFO] [1694118463.408046250]: Camera model: fx = 662.989, fy = 665.07, cx = 327.432, cy = 220.266
[ WARN] [1694118463.862154903]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1694118464.863088387]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1694118465.863867381]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1694118466.866712606]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1694118467.869086315]: "camera_odom_frame" passed to lookupTransform argument target_frame does not exist.
```
What i have tried: i see /apriltags_to_mavros/source_frame_id: camera, i have tried changeing "camera" to "usb_cam" from /usb_cam/camera_frame_id: usb_cam. I also tried usb_cam_frame and that didnt work either.

I noticed the following warnings, and tried to edit the cpp code but even when changing the cpp code this does not change.
``
[ WARN] [1694118460.817885468]: Using default target_frame_id: /camera_odom_frame
[ WARN] [1694118460.825135741]: Using default source_frame_id: /camera_link
``

Im not exactly sure what im not understanding correctly, any help would be greatly appreciated.

Thank you in advance!

Dawgomatic, 2023-09-12:
I got /mavros/vision_pose/pose and /tf to give me data by changing vision_to_mavros.cpp lines 32 and 34 to the correct naming conventions. then, deleting the build and devel folders in my work space and rebuilding using catking_make. I am now running into trouble getting the scene to show in RVIZ, when i add the topic Pose under /mavros/vision_pose/pose topic. i get the following:
```
transform[sender=unknown_publisher]
For frame [a3_bundle]: No transform to fixed frame[map]. TF error:[could not find a connection between 'map' and 'a3_bundle' because they are bit part of the same tree. Tf has two or more unconnected trees.]

``` 

Any guidance here?

thien94, 2023-09-13:
Hi @Dawgomatic , sorry for the late reply.

For the RVIZ issue, you can change the name of `fixed frame` in the `Global Options` tab to `a3_bundle`. Then the TF tree should show up in the visualization.

Dawgomatic, 2023-09-13:
That did the trick! thank you so much for the help!

closed 2023-09-14 by thien94